nebkat/esp32-xbee #1: flashing process

aortner, 2019-11-16:
hi. i have got an eary version from ardusimple.

i did follow your firmware update step by step - it did not work at me

solution: i need to press boot some times to start the flashing...

nebkat, 2019-11-18:
Hi Andreas, did you follow the process to enter download mode by holding BOOT button + pressing RESET, and then have to press BOOT button on its own after this? Or do you mean just this initial button combination?

aortner, 2019-11-19:
oh i see:

first: boot + reset
the boot button again ..

you should add this to your docs.. :-)

nebkat, 2019-11-20:
Hmm, the second boot press shouldn't actually have any effect. If you just hold BOOT + press and release RESET, does it work correctly?

closed 2019-12-04 by nebkat