nebkat/esp32-xbee #2: Compile with PlatformIO

hagre, 2019-11-20:
Hello, I am trying to compile with PlatformIO and espidf framework, but i still have lots of errors.
I moved the /main files to the /src folder and some other small adjustments, but it is still not working.
Help needed! Has anyone a good guideline for this?

nebkat, 2019-11-20:
Could you attach the error logs? I'm not sure how ESP-IDF works for PlatformIO, but you can also use the standard ESP-IDF build system https://docs.espressif.com/projects/esp-idf/en/latest/get-started/

closed 2019-12-04 by nebkat

hagre, 2020-01-03:
Sorry for may late replay,
Thank you for your suggestion, I am using the standard ESP-IDF build system now. No luck and too much manual converting work for PlatformIO and VScode (problems with differnet folder structure and config files....)

arnolde, 2020-09-22:
I only use platformio for all my projects and I would appreciate if anyone has already converted their source and would share it (or a guide howto).