nebkat/esp32-xbee #4: boot loop

retkon, 2019-12-14:
I have a boot loop with version 0.4.2, reflashing doesn't help, full reset doesn't work.

Terminal Output:
_$PESP,WIFI,AP,SSID,ESP_XBee_D4A4F5,V,O*28
$PESP,INIT,START,v0.4.2,SOFTWARE*03
$PESP,WIFI,AP,SSID,ESP_XBee_D4A4F5,V,O*28
$PESP,INIT,START,v0.4.2,SOFTWARE*03
$PESP,WIFI,AP,SSID,ESP_XBee_D4A4F5,V,O*28
$PESP,INIT,START,v0.4.2,SOFTWARE*03
$PESP,WIFI,AP,SSID,ESP_XBee_D4A4F5,V,O*28
$PESP,INIT,START,v0.4.2,SOFTWARE*03_

nebkat, 2019-12-18:
Hi @retkon,

Does these lines get spammed very quickly or is there a delay between each set of $PESP,INIT lines?

retkon, 2019-12-18:
Hi @nebkat,

Between each double line is 1.5s pause.

nebkat, 2019-12-20:
Hi @retkon,

Could you try the latest firmware release? I have added a delay to allow you to perform the full reset.

retkon, 2019-12-21:
Hi @nebkat 
Thanks for the new Firmware
Unfortunately it doesn't help. 
Making a full reset is still not possible 
and the output is the same, 
except there is a little longer delay between and now a third line is in output:

$PESP,INIT,START,v0.4.3,SOFTWARE*02
(delay 2sec)
$PESP,WIFI,AP,SSID,ESP_XBee_D4A4F5,V,O*28
$PESP,WIFI,STA,CONNECTING,MyWLAN01_AP,P*67
(delay 2sec)
$PESP,INIT,START,v0.4.3,SOFTWARE*02
(delay 2sec)
$PESP,WIFI,AP,SSID,ESP_XBee_D4A4F5,V,O*28
$PESP,WIFI,STA,CONNECTING,MyWLAN01_AP,P*67

Any suggestions ?

nebkat, 2019-12-21:
I will try to prepare a special build for you tonight to allow you to reset, but until then, are you using a good power source? We have noted some problems when using long USB cables or inadequate USB power supplies - perhaps you could try plugging the cable into a wall charger if you have not done so already?

retkon, 2019-12-21:
Try'd now a wall charger (2.5A) with with short cable, no difference. 
But it was worth a try.

retkon, 2020-01-06:
Happy New Year ;-)

> I will try to prepare a special build for you tonight to allow you to reset,

Have you already finished the special build for me ?

nebkat, 2020-01-07:
Happy new year @retkon!

Apologies for the delay, could you try to flash the latest firmware v0.4.4, and if that does not help then flash it again but replace with this esp32-xbee.bin. This adds an 8 second delay after boot so you can hold the boot button long enough for the full reset to occur. Then if all is working as expected you can go back to the original esp32-xbee.bin.

[esp32-xbee.bin.zip](https://github.com/nebkat/esp32-xbee/files/4031696/esp32-xbee.bin.zip)

retkon, 2020-01-10:
Thank you for the special version. But i was not able to reset to factory setting with pressing the button 5sek (or more), it had no effect. 
But after many trys i was able to erase the ESP32 completly an reflash it with the latest version.
Now the Xbee ist working :-).

closed 2020-01-10 by retkon