nebkat/esp32-xbee #5: Long term stability

AndrewR-L, 2019-12-22:
I have experienced a number of issues attempting to run v0.4.2 simultaneously as an NTRIP server and TCP server which may be related. To rule out power fluctuations power is supplied to simplertk2b simultaneously and independently via both USB sockets.

1. when running continuously esp32-xbee disconnects both NTRIP and TCP every 20865 seconds. This results in a loss of coverage eg because RTKLIB STRSVR which connects to esp-xbee socket takes 10 seconds to timeout retrying and a further 10 seconds after reconnecting to get up and running with useful data.

2. esp32-xbee is also experiencing "panic" resets. I have attached logs and core_dumps of a couple of recent occasions. In the second instance I have included some heap_info outputs to see if this was caused by stack overflow. In each instance you can see effect no.1 happening 5hrs 50 mins after the TCP socket was first set up.

3. after a few days esp32-xbee seems to lose wifi connectivity needing sometimes a full reset (>5 secs on boot button) to get running again. Because of issues no.1 and no.2 above I've not investigated whether this is a problem with my wifi or with xbee.  

Are any of thee issues addressed in the new v0.4.3 just posted?

[core_dumpand log2.zip](https://github.com/nebkat/esp32-xbee/files/3992924/core_dumpand.log2.zip)
[core_dumpandlog1.zip](https://github.com/nebkat/esp32-xbee/files/3992925/core_dumpandlog1.zip)

PS I am not sure what purpose uart_log(buffer, n) has in log.c in the current implementation but I wonder whether it should be called regardless of whether xRingbufferSend returns an error when the ring buffer gets full.

AndrewR-L, 2019-12-22:
UPDATE:

Now it is disconnecting and reconnecting even more frequently c4hours and then c2hours see extended log ...
[espxbeelogpanicagaincontinued.txt](https://github.com/nebkat/esp32-xbee/files/3993142/espxbeelogpanicagaincontinued.txt)

AndrewR-L, 2019-12-23:
UPDATE:

Just in case TWO output streams was a bit much I turned off the socket server, but still getting panic resets every few hours. Does the core_dump indicate the cause of the "panic"?

[core_dumpandlog3.zip](https://github.com/nebkat/esp32-xbee/files/3994184/core_dumpandlog3.zip)

AndrewR-L, 2019-12-23:
UPDATE:

Running with NO server and NO client enabled seems (so far) to have cured the panic resets but still periodically "httpd_txrx: httpd_sock_err: error in recv : 113" messages are coming up (also errno 104) which is what has been happening all along each time esp32-xbee disconnected the TCP Server socket and NTRIP so is probably not a coincidence.

[espxbeelognoop.txt](https://github.com/nebkat/esp32-xbee/files/3995699/espxbeelognoop.txt)

Unfortunately it doesn't say which socket (!) so I guess it is socket 80 listening for www.

No idea why errors should be occurring, but I have checked it is not a local network issue because disconnecting Wifi produces many different logs.

Question: Is there something in the httpd_sock_err recovery code which causes ALL sockets to be reset? Or something which is corrupting all the handles on the stack? If so that could explain issue no.1.

PS Now going to test with just socket server enabled to see what happens.

nebkat, 2019-12-29:
Hi Andrew,
Thank you for all your research! I will be back from vacation tomorrow and will look through the logs to figure out a solution. Let me know if you have any new information.

AndrewR-L, 2019-12-29:
UPDATE:

Running with just the socket server enabled still results in panic resets.

Here is a log and 2 core_dumps.

[core_dumpandlog4.zip](https://github.com/nebkat/esp32-xbee/files/4008318/core_dumpandlog4.zip)

CONCLUSION

1. "Panic" resets

These occur when either server is enabled or both. The only stable solution seems to be when no output is enabled (!). [NB I have not tried it long term with clients or caster enabled]

2. Socket disconnections

I have noticed that rtk2go.com lists mountpoint ne36 as using "NTRIP_ESP32_XBee_Ntrip_Server/1.0" but achieving uninterrupted periods of 23hrs+ so I think this is my issue not a feature of the code.

To check that I swapped out the esp-xbee for a RS232 xbee module hard wired to an HP thin client (windows XP running STRSVR in the background) connected to the same wifi and this too loses connection to rtk2go at random intervals min 5 minutes max 7hrs+ so I think this is a network issue - maybe a feature of CGNAT used by EE - maybe the intervals are now random rather than the predictable 20865 seconds because it depends on total network traffic. Perhaps these disconnections and reconnections are what is upsetting the code causing the panics?

With the wired RS232 connection the solution is reasonably robust as the stream recovers in 13 seconds so this is less of an issue than the esp32 panic resets which take longer or wifi problems which are terminal.

3. Losing wifi connectivity

This is less frequent but the most inconvenient because it requires manual intervention. I have no clue what is going on. The logs above include a couple of instances with different symptoms.

Thanks for offering to look into this - I hope this gives you enough to narrow down the cause.

AndrewR-L, 2020-01-03:
NOTE

In other circumstances esp32-xbee running the same version NTRIP server can do longterm continuous - see extract from rtk2go.com status report below. I guess there is just some limitation in my setup which it does not tolerate well.

**Base Station Mount Point Details: ne36**
Station details for stream:    ne36 
  Located at:   Donavon, Canada
  NTRIP Agent:   NTRIP_ESP32_XBee_Ntrip_Server/1.0
...
Recent Uptime Stats and Plots
   Last Restart: 3 Days 14:44 Up(21/46) (3 Days 14:44 (HH:MM)  100.0% overall) is 46th connection

nebkat, 2020-01-05:
For some reason all of your core dumps are giving me this error so I can't get anything useful out of them, you didn't happen to compile your own v0.4.2 binaries? The ELF file probably needs to match the core dump.

```/opt/esp-idf-sdk/components/espcoredump/espcoredump.py dbg_corefile -t raw -c Downloads/esp32_xbee_core_dump_2019-12-26_23_54_02.bin -r ~/Downloads/esp32_rom.elf ~/Downloads/esp32-xbee\ \(2\).elf
espcoredump.py v0.4-dev
WARNING: Skip task's (3ffc6690) stack 456 bytes @ 0x3ffc6690. (Reason: Can not add overlapping region [3ffc6690..3ffc6857] to ELF file. Conflict with existing [3ffc6690..3ffc67f3].)
ERROR: Growing up stacks are not supported for now!
ERROR: Failed to create corefile!
```
The httpd_txrx errors are `ECONNRESET 104 Connection reset by peer` and `EHOSTUNREACH 113 No route to host` which don't seem too serious.

WiFi and panics are still a mystery to me from those logs unfortunately. I'll release a new version today with some minor fixes, would be good to see if there is any improvement.

Thanks again for all the research!

AndrewR-L, 2020-01-06:
No, I am using the binaries which came pre-loaded from Ardusimple (the only reason I looked at the code was to see if I could get any clue as to what was going wrong).

But googling for that error found someone who had solved the growing up stacks error ...

_"Got this figured out. If I remove the first four bytes (the magic number) it all works fine! Looks like the flash loader removes these automatically and the raw loader doesn't expect them."_

Does this help get any clues from the core_dumps?

Otherwise I'll look out for your new binaries (0.4.4?) and let you know when I get a chance to swap the ESPXbee back in.

AndrewR-L, 2020-01-07:
Tried uploading v0.4.4 as per the guide. I got the LED to stop flashing but the upload always FAILed as below (both @ 115200 and 460800 baud).

I am using POWER+XBEE port on SimpleRTK2B to upload. How is this wired? Is it necessary to disable all UART2 output to avoid conflict (I didn't do so because the instructions didn't ask me to do that).

Does the uploader produce any kind of log to indicate why the fail, or have I made an obvious mistake?

[image]

nebkat, 2020-01-08:
Yes you are correct, I've updated the guide with a warning about that.

That is very unusual... were you ever able to successfully complete an update? The flash size doesn't matter. Could you try flash this? Same locations as above, just no www.bin.
[blink.zip](https://github.com/nebkat/esp32-xbee/files/4035691/blink.zip)

AndrewR-L, 2020-01-08:
No I never succeeded. Got the green FINISH box several times as above but no change in the XBee, Couldn't find any kind of a log to see what's what.  Will try blink.zip

AndrewR-L, 2020-01-08:
Found a console window in the background. It says …

```
======
CONNECT BAUD: 115200
============
....Efuse CONSOLE_DEBUG_DISABLE is already burned.
Uploading stub...
Running stub...
Stub running...

 is stub and send flash finish
```

nebkat, 2020-01-08:
This alone doesn't seem incorrect. Could it be because you have not checked the boxes beside the bin files in the flasher tool?

AndrewR-L, 2020-01-08:
It was! v0.4.4 loaded now thanks. Will see how it runs.

AndrewR-L, 2020-01-09:
I am sorry but it is still panic resetting (at least twice already).

See attached  (also some new messages from ntrip_server_task)

Are these core_dumps readable/useful?

[espxbee-044.zip](https://github.com/nebkat/esp32-xbee/files/4039499/espxbee-044.zip)

AndrewR-L, 2020-01-11:
Still getting odd behaviour (see log) and panic resets (do you need any more core dumps?)

I don't know how to investigate the cause of the panic_resets. but I suspect much of the odd behaviour is because the wifi router connects over a 4G link (EE) rather that copper/fibre (I guess 4G is more common in rovers than bases). So I have been looking at ntrip_server.c to see if it could be made less sensitive to temporary dropouts. I am not familiar with ESP/FreeRTOS but I assume it already includes generous retries/timeouts in socket write() etc.

BUT I have noticed that when ANY error occurs relaying a UART stream to the TCP socket you call destroy_socket straight away (setting sock=-1 etc) and any further UART stream is dumped (and server_keep_alive is no longer zeroed). BUT the keep alive loop does not know that and keeps looping for 10s and only eventually breaks out when it tries to send _newline_ to sock no -1.

Do we need that 10s delay in this instance? Could we change the keep alive loop from `while(true)` to `while(sock!=-1)` so that it drops out after <1 second if the socket has gone and enters the reconnection phase (via the retry_delay in the outer while(true) loop which stops us hamering the caster anyway).

For that matter once streaming has begun would we ever want to send _newline_ keep alive characters in the middle of a binary RTCM stream (and if ever it was triggered would that confuse the caster?).

Also I have noticed that the uart_handler starts relaying as soon as the socket is connected (ie whenever sock!=-1) and does not wait for the NTRIP logon dialog to complete! Probably it is contrary to NTRIP protocol and may confuse the login if things happen in the wrong order?

Log to date attached with added comments status reports etc.

AndrewR-L, 2020-01-11:
[espxbee-044-continued.txt](https://github.com/nebkat/esp32-xbee/files/4049526/espxbee-044-continued.txt)

DavidKelleySCSC, 2020-01-11:
I have been watching this thread ever since the RTK2go.com NTRIP Caster (use-snip.com) was mentioned on the site.  A couple of basic NTRIP design issues to be aware of are mentioned below as the code is improved.

> For that matter once streaming has begun would we ever want to send newline keep alive
> characters in the middle of a binary RTCM stream (and if ever it was triggered would that 
> confuse the caster?).

The presence of a newline or return character is not allowed in an RTCM 3.x message stream and will cause message loss issues with Casters and Client devices.  Do not do this.  [Aside some older RTCM 2.x message stream sources do contain \r\n but this is also incorrect, some NTRIP Client devices can cope, others cannot.] If you are looking at the raw binary data, the first char of most RTCM3 messages can be detected by 0xD3. 

> and does not wait for the NTRIP logon dialog to complete! Probably it is contrary to 
> NTRIP protocol and may confuse the login if things happen in the wrong order?

Ideally the 200 Ok return is sent back by the Caster to the connecting socket (the NTRIP Client or Server), and then binary data transmission begins.  It may be of value to know the original NTRIP protocol was derived in part from the simple 'shoutcast' protocol.  More then a few NTRIP Clients do not strictly follow this, so the SNIP Caster (like many other Caster designs) can cope either way.   The 'error reply' for NTRIP (Rev1) is extremely simple; if anything goes wrong with the initial request, the Caster sends the 'caster table' back and disconnects.  In the above case, were \r\n to be found in the data stream, the NTRIP Clients using that data would not be able to reliably decode the messages. 

The only other general remark I think needs to be made is that it is best that any NTRIP Server does not connect to the remote NTRIP Caster unless it in fact has some message content to be sent.  Otherwise, (using SNIP as a typical example), the connection is made, no further data is ever sent, the Caster notices this and after some delay (10~15 seconds) disconnects the connection.  Then the NTRIP Server connects again and this cycle continues forever (in the case of SNIP certain abuse logic will in time ban the IP from connecting for a variable period of time).   The takeaway is that this code should confirm there is data to be sent before establishing the connection in the first place.  [Aside, RTKLIB has this issue as well, although the RTK Explorer version has added a test for the presence of data.]

AndrewR-L, 2020-01-13:
That is very illuminating thank you.

In the context of SimpleRTK2b with ZED-F9P I think in practice ...

1. Once streaming has started the _newline_ in the 10sec timeout loop will never be triggered if the RTCM stream is coming in every second. In this scenario perhaps it is sufficient to keep the loop but have it exit when it times out (or earlier if sock becomes -1) because the caster is going to disconnect "after some delay (10~15 seconds)" anyway.

2. I don't use it but ZED-F9P has a survey-in mode with bases which can be relocated and I guess the UART may be silent for a few minutes while that happens(?) and perhaps this is what the _newline_ keep alive was intended for (ie between text-mode connection and the binary starting). If so I can see the logic but as @DavidKelleySCSC points out probably not best practice also I fear a few minutes of _newline_ may confuse his SNIP's clever auto-set caster table feature if there is no real data to parse in the first couple of minutes of connection.

So perhaps a way forward is -
(a) for _ntrip_server_task_ to pause before opening the socket and wait until there is some uart data.

(b) for _ntrip_server_uart_handler_ to dump data unless _ntrip_server_task_ has set a flag - which it would do just after sending the "successfully connected" log and reset just prior the to the "disconnected" log. If necessary also have the handler set another flag if data is received while sock=-1 so as to trigger opening the socket as in (a).

(c) trim the keep alive loop thus

```
while (sock != -1) {

            if (server_keep_alive >= NTRIP_KEEP_ALIVE_THRESHOLD) break;

            server_keep_alive += NTRIP_KEEP_ALIVE_THRESHOLD / 10;

            vTaskDelay(pdMS_TO_TICKS(NTRIP_KEEP_ALIVE_THRESHOLD / 10));

        }
```

NB brief disconnections and re connections although preferably avoided don't seem to upset the rover's lock (but I haven't confirmed that by having it drive a tractor yet) just longer interruptions like the panic resets or wifi issues.

AndrewR-L, 2020-01-13:
[esp32_xbee_v0.4.4_core_dump_2020-01-13_12_05_13_79465b9f.zip](https://github.com/nebkat/esp32-xbee/files/4053467/esp32_xbee_v0.4.4_core_dump_2020-01-13_12_05_13_79465b9f.zip)

AndrewR-L, 2020-01-19:
Today's panic reset. Is it possible to determine the cause of the panic and/or which task was active when it happened to see if there is a pattern?

[esp32_xbee_v0.4.4_core_dump_2020-01-19_10_02_49_79465b9f.zip](https://github.com/nebkat/esp32-xbee/files/4082491/esp32_xbee_v0.4.4_core_dump_2020-01-19_10_02_49_79465b9f.zip)

AndrewR-L, 2020-01-23:
Had achieved over 19hours uninterrupted connection to rtk2go then another panic reset ,,,
[esp32_xbee_v0.4.4_core_dump_2020-01-23_19_50_53_79465b9f.zip](https://github.com/nebkat/esp32-xbee/files/4105175/esp32_xbee_v0.4.4_core_dump_2020-01-23_19_50_53_79465b9f.zip)

nebkat, 2020-01-28:
@DavidKelleySCSC Thank you for this information. I have implemented a system now similar to @AndrewR-L's suggestion to deal with this (bb131ab).

Before the initial connection it will wait for some data on UART, then once connected there will be a task similar to the previous keep alive that waits for 10 seconds of no data to (un)set the data ready flag. It will never disconnect from the caster itself, but if the caster chooses to kick the server it will not attempt to reconnect again without new data being sent.

@AndrewR-L Unfortunately all of these core dumps continue to give me the same error and still I don't get the same issue with my own core dumps. I'm going to increase the core dump partition size just in case but from what I've read if the partition size was an issue the core dump simply wouldn't be saved.

> BUT I have noticed that when ANY error occurs relaying a UART stream to the TCP socket you call destroy_socket straight away (setting sock=-1 etc) and any further UART stream is dumped (and [...]

> Also I have noticed that the uart_handler starts relaying as soon as the socket is connected (ie whenever sock!=-1) and does not wait for the NTRIP logon dialog to complete! Probably it is contrary to NTRIP protocol and may confuse the login if things happen in the wrong order?

Implemented in server/client (c3dc326). Thanks again for all your investigation! Hopefully this will solve some of the panics.

nebkat, 2020-01-29:
Still haven't come across anything obvious that would have caused all the panics but there were a lot of small fixes in the last release so it would be great if you could let me know how that performs for you.

AndrewR-L, 2020-01-29:
Great, thanks, I'll try it out and let you know. The last time v0.4.4 reset itself out of the blue it said "reset reason: INTERRUPT_WATCHDOG" which was a new one to me. 

It is a shame you couldn't read the core_dumps - I can't think what I am doing different. Hopefully there will be no more "panic resets" with v0.5. What does an ESP "panic" indicate generally - is that a hardware thing (eg overheating) or a software thing (eg memory issues)?

PS thanks for picking up the wifi suggested enhancements.

nebkat, 2020-01-29:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html

This has the full list, its almost certainly a software issue. ESP32 has the added complexity of having two cores so usually problems are caused by memory issues.

DavidKelleySCSC, 2020-01-29:
Gentlemen, sounds like good progress, congratulations.  If AndrewR-L can tell me the mountPt name that is being used, I will set up some extra monitoring on the RTK2go.com SNIP NTRIP Caster to record the connection times and help you validate things.

AndrewR-L, 2020-01-29:
@DavidKelleySCSC that is very kind. It is on NR152QB, still using the old firmware for now, but I'll swap it to the new issue at the weekend.

On that topic was rtk2go.com supposed to be uncontactable for a few hours leading up to around 15:00 (UTC) last Saturday (25th) or was that an issue this end? Either way it demonstrated that the server did gracefully escalate its retry interval when it couldn't contact the mountpoint 1sec - 2 sec - 5 sec - 10 sec - 15 sec - 30 sec and so on eventually ending up with hourly retries.

DavidKelleySCSC, 2020-01-30:
Got it recording now in case detailed records are needed.  You can always get a real time status report (when it is connected) with: http://rtk2go.com:2101/SNIP::MOUNTPT?NAME=NR152QB

Here is a quick plot from SNIP shpowing the past 4 weeks, you can see a fair number of up/dwn events as well as the two recent periods that RTK2go was offline. [One came form a Microsoft bug check reboot, not sure why. The other other was quite odd, all IPs on the machine appeared open from the console yet no traffic would flow. We have seen this twice on Window7 machine in the past] (Click to see larger)
[image]

What you really cannot see from the above is that there are several re-connect events that seemed to me to take way too long (minutes rather than seconds),   I will keep an eye on these and report what is seen.

DavidKelleySCSC, 2020-01-31:
Now ~24 hours later, no sign of odd drop out and restarts as well.  While it is too early to say for sure, looks much improved.

AndrewR-L, 2020-02-05:
Apologies I didn't get the upgrade to v0.5 done until today.

That outage on 30 Jan was not the reprogramming but rather wifi connectivity disappeared. Not only did it lose the connection to the site wifi, but also it was not broadcasting from its own internal AP. Xbee seemed to be none the wiser - still flashing the LED to indicate NTRIP server and caster function - just not reaching the outside world.

So fingers crossed from here.

BTW when rtk2go.com caster was offline recently because no traffic would flow, the Rtk2go Front Page  www.rtk2go.com was also unavailable (you know the page one nobody ever reads but does have the "Current Status" message on it). Perhaps it is hosted on the same Windows 7 machine?

AndrewR-L, 2020-02-07:
This is interesting. The new "No data received by UART in 10 seconds" log came up after several hours of connection, which is a surprise because I had assumed that once the ZED-F9P started streaming RTCM is did so every second without fail. Perhaps this has always been the cause of the briefer disconnects and reconnects (not the panics etc) and maybe that had after all been confounded by that newline previously sent on timeout.

Curiously despite saying "will not reconnect to caster if disconnected" when the caster eventually disconnected a reconnection was attempted just 0.004 seconds later without waiting for UART data (and the connection did succeed).

```
I (08:45:47.081) MAIN: Synced time from SNTP
W (08:57:15.356) NTRIP_SERVER: No data received by UART in 10 seconds, will not reconnect to caster if disconnected
W (08:57:25.647) NTRIP_SERVER: Disconnected from rtk2go.com:2101/NR152QB
I (08:57:25.651) NTRIP_SERVER: Connecting to rtk2go.com:2101/NR152QB
I (08:57:29.053) NTRIP_SERVER: Successfully connected to rtk2go.com:2101/NR152QB
I (09:45:46.279) MAIN: Synced time from SNTP
```
So did UART data really only become available in the split second after rtk2go gave up waiting or was it available sooner but the UART had stopped listening after 08:57:15.356 so didn't serve the data which would otherwise have avoided the rtk2go connection starving to death?

NB Full log attached
[V5log1.txt](https://github.com/nebkat/esp32-xbee/files/4171257/V5log1.txt)

nebkat, 2020-02-07:
There was no message to indicate that UART data had been sent since the last warning but I've added that now to simplify debugging.

```
W (16:18:18.951) NTRIP_SERVER: No data received by UART in 10 seconds, will not reconnect to caster if disconnected
I (16:18:34.121) NTRIP_SERVER: Data received by UART, will now reconnect to caster if disconnected
W (16:18:44.951) NTRIP_SERVER: No data received by UART in 10 seconds, will not reconnect to caster if disconnected
I (16:19:15.021) NTRIP_SERVER: Data received by UART, will now reconnect to caster if disconnected
```

That said, it is unusual that the warning came up after many hours. I have had a slightly different issue where the warning is printed before the initial connection is ever made, i.e. the sleep task is being resumed somehow.

I'll be adding a web page to view the UART data coming in and out soon so we'll be able to debug that problem anyway.

AndrewR-L, 2020-02-07:
I'll look out for that, thanks.

Just curious, in ntrip_server.c ...

```
        // Connected
        xEventGroupSetBits(server_event_group, CASTER_READY_BIT);

        // Await disconnect from UART handler
        vTaskSuspend(NULL);

        // Disconnected
        xEventGroupSetBits(server_event_group, CASTER_READY_BIT);
```
... is it correct that the CASTER_READY_BIT is set when disconnected or should it be reset instead?

AndrewR-L, 2020-02-08:
That warning came up again twice after being up continuously for over 27hours

The first time at 12:14:06 the UART data clearly resumed soon enough to prevent rtk2go dropping out.
The second time at 12:15:36 it seems that when the UART data resumed it found it could not write to the socket because rtk2go had gone already so it closed the socket and triggered a reconnection. I realise now it was wrong to expect a "Waiting for UART input to connect to caster" because it was in fact the PRESENCE of UART input which had triggered the connection!

So the server does seem to be doing all you could ask of it - serving the data whenever it is available. (I will have to investigate what causes these pauses in the RTCM stream - does SimpleRTK2b stop sending or does Xbee stop receiving?)

[image]

AndrewR-L, 2020-02-10:
I tested disconnecting the antenna even that did not stop the RTCM stream (message 1230 continued even after the others petered out) so these gaps are not a sky issue.

But I have discovered my board shipped with v1.11 of the ZED-F9P firmware so I will try v1.12 to see if that helps and if not log the ZED-F9P output to see whether the pauses are because it has stopped talking or whether XBee has stopped listening.

nebkat, 2020-02-10:
> ... is it correct that the CASTER_READY_BIT is set when disconnected or should it be reset instead?
Looks like that should indeed be xEventGroupClearBits. If you find any more like that feel free to comment directly on the code (to save you time copying!).

Are you now running the latest v0.5.1?

AndrewR-L, 2020-02-11:
I'm new to Git/GitHub so I was not entirely sure how to do that, but I have just had a practice (and it seemed to work!)

No not yet, still v1.11 and v0.5 but for now they do seem to be behaving. Intend to upload both tonight (and start logging ZED-F9P)

nebkat, 2020-02-12:
If it would help I could also offer you access to my SNIP caster, then I could take a look at the logs to see exactly what is going on.

AndrewR-L, 2020-02-12:
Thank you that may help but I already got it set up (with FW 1.12 and 5.1) simultaneously ...

(a) XBee serving directly to rtk2go 
(b) U-centre recording ZED-F9P USB output (with UBX debugging turned on) for Ardusimple
(c) RTKLIB STRSVR client connected to XBee caster both logging the RTCM to a file.

... so as to be able to compare all 3 whenever the XBee log shows a hiccup.

We'll see how it goes but I bet it won't go wrong while being watched so intently!

AndrewR-L, 2020-02-13:
```
W (12:20:48.127) NTRIP_SERVER: No data received by UART in 10 seconds, will not reconnect to caster if disconnected
I (12:20:52.322) NTRIP_SERVER: Data received by UART, will now reconnect to caster if disconnected
```

(a) rtk2go did not disconnect (data resumed sufficiently quickly) but counted this as a restart.

(c) strsvr stopped receiving from the XBee local caster at the same time.

(b) replaying the ublox log shows no obvious hiccup in the UBX stream, and the specimen RTCM and NMEA message sent to the USB show up at the proper times throughout and the UBX MON-IO suggests that ZED-F9P was outputting data to UART2 all the time ...

- from 12:20:39 to 12:20:48 it shows 8883 bytes output to UART2

- from 12:20:49 to 12:20:58 it shows 8872 bytes output to UART2

- from 12:20:59 to 12:21:08 it shows 8973 bytes output to UART2

I can check other UBX messages but this is pretty consistent so it seems that it is XBee which stops listening (while F9P keeps talking). What could cause that?

I see you now have v0.5.2, perhaps I should try that.

AndrewR-L, 2020-02-18:
Still getting the brief dropouts of around 10s sec, but ranging from 5secs to 15+ seconds but there seems no issue at the SimpleRTK2B end.

I have noticed that RTCM says

_The Variable Length Messages are those defined in Chapter 3. If the data link requires short
messages in order to maintain a continuous stream of data, the message length may be set to "0",
providing a "filler" message of 48 bits in length, because the data message length will be zero._

so I am thinking that maybe the time has come to put back the keep-alive transmission (but using this binary filler messge rather than newline) and see if the rover is at all bothered by these blips. I think maybe upto 30 seconds should serve the purpose without annoying casters such as rtk2go by hanging on for too long.

I am of course assuming that a nul message is sufficient to reset a caster's timeout.

AFAIK mixing UBX, RTCM and even NMEA does not confuse a UBX reader (at least U-Centre copes with this) so I am thinking that this will serve the purpose even if raw data is streamed to the caster.

I will push a suggestion to do this

AndrewR-L, 2020-02-22:
Right, so in the absence of keep alive and as suggested in response to pull request #14 I am digging further into what is causing the absences of UART.

So I have bought a TTL to USB adapter and connected it to SimpleRTK2b RX2 which is ZED-F9P UART2 output and am recording ...

1. with U-Centre from the SimpleRTK2b USB the full debugging log which includes MON-IO showing no of bytes output to UART2.

2. with STRSVR from UART2 the RTCM steam output to XBEE with timestamping

3. periodically the XBee/log messages

If that doesn't tell us where the breakdown in communication is arising I don't know what will.

Is anyone else experiencing these blips in the RTCM stream?

AndrewR-L, 2020-02-23:
It seems that this is an XBee issue not a fault in Simplertk2b or ZED-F9p.

Here is what happened after being up with no issue for over 20 hours

```
I (00:00:06.771) NTRIP_SERVER: Successfully connected to rtk2go.com:2101/NR152QB
I (19:17:08.465) MAIN: Synced time from SNTP
I (20:17:08.602) MAIN: Synced time from SNTP
I (21:17:08.920) MAIN: Synced time from SNTP
I (22:17:09.048) MAIN: Synced time from SNTP
I (23:17:10.073) MAIN: Synced time from SNTP
I (00:17:09.528) MAIN: Synced time from SNTP
I (01:17:10.503) MAIN: Synced time from SNTP
E (02:08:51.403) NTRIP_CASTER: ../main/ntrip_caster.c:147 (ntrip_caster_task): Client did not send GET request
I (02:17:10.677) MAIN: Synced time from SNTP
I (03:17:10.938) MAIN: Synced time from SNTP
I (04:17:11.054) MAIN: Synced time from SNTP
I (05:17:11.368) MAIN: Synced time from SNTP
I (06:17:11.572) MAIN: Synced time from SNTP
I (07:17:11.695) MAIN: Synced time from SNTP
I (08:17:11.907) MAIN: Synced time from SNTP
I (09:17:12.108) MAIN: Synced time from SNTP
I (10:17:12.316) MAIN: Synced time from SNTP
I (11:17:12.524) MAIN: Synced time from SNTP
I (12:17:12.736) MAIN: Synced time from SNTP
I (13:17:12.943) MAIN: Synced time from SNTP
I (14:17:13.132) MAIN: Synced time from SNTP
I (15:17:13.246) MAIN: Synced time from SNTP
W (15:54:55.976) httpd_txrx: httpd_sock_err: error in recv : 104
W (15:55:06.738) NTRIP_SERVER: No data received by UART in 10 seconds, will not reconnect to caster if disconnected
I (15:55:13.778) NTRIP_SERVER: Data received by UART, will now reconnect to caster if disconnected
```
and rtk2go.com reported the 15:55 event as a "restart" not a "reconnection".

BUT (see extract of .obs file created by RTKCONV from the captured UART2 log the output from ZED-F9P UART2 didn't miss a beat and kept coming every second, which confirms what the UBX MON-IO message is reporting.

[uart2-xbee.log](https://github.com/nebkat/esp32-xbee/files/4241904/uart2-xbee.log)

I was also logging the output from the XBee internal caster locally with STRSVR and there was a gap in the data of 36 seconds at this time (probably the 17 second gap in the UART2 data reported in the Xbee log got stretched due to timeouts in the caster and/or strsvr).

Coincidentally the UART2 timeout occured after a http 104 message. I had noticed that previously (see further up this thread) but recently the UART2 timeout had been occuring without it.

Are there events such as this which might turn off interrupts for a while or hog the ESP32 cpu at a higher priority than the UART2 task?

PS The issue at 02:08 did not affect the ntrip server but for some reason STRSVR didn't receive anything from the XBee ntrip caster for >10 seconds and timed out. Maybe a similar issue prevented the caster task running for a while?

AndrewR-L, 2020-02-25:
First panic reset with v0.5.2.

Is this readable now?
[esp32_xbee_v0.5.2_core_dump_dd4176_2020-02-25_11_28_3.bin.zip](https://github.com/nebkat/esp32-xbee/files/4249743/esp32_xbee_v0.5.2_core_dump_dd4176_2020-02-25_11_28_3.bin.zip)

nebkat, 2020-02-25:
@AndrewR-L 
That one was actually readable now, but nothing showed up that explained the panic. Only thing related to esp32-xbee code was:

```
#0  0x40094a14 in xQueueReceiveFromISR (xQueue=0x3ffc5188, pvBuffer=0x0, pxHigherPriorityTaskWoken=0x3ffc51d0) at /opt/esp-idf-sdk/components/freertos/queue.c:1649
#1  0x400877aa in rtc_time_get () at /opt/esp-idf-sdk/components/soc/esp32/rtc_time.c:149
#2  0x40087c40 in rtc_init (cfg=...) at /opt/esp-idf-sdk/components/soc/esp32/rtc_init.c:78
#3  0x400e7538 in nvs::Lock::Lock (this=<synthetic pointer>) at /opt/esp-idf-sdk/components/nvs_flash/src/nvs_platform.hpp:30
#4  nvs_set<int> (c_handle=0, key=0x3ffc3e0c <error: Cannot access memory at address 0x3ffc3e0c>, value=4096) at /opt/esp-idf-sdk/components/nvs_flash/src/nvs_api.cpp:279
#5  0x400dacde in config_get_handler (req=0x3ffa7544) at ../main/web_server.c:456
#6  0x40093dfd in _frxt_int_enter () at /opt/esp-idf-sdk/components/freertos/xtensa/portasm.S:134
```

The Recv 104 Connection Reset by Peer seems suspicious but there is no reason that would cause UART input to stop, and it might just be a symptom of the problem. 

@DavidKelleySCSC 
I've noticed SNIP seems to be returning a double ICY 200 OK for some requests, is this the expected behavior or a bug?

DavidKelleySCSC, 2020-02-28:
> I've noticed SNIP seems to be returning a double ICY 200 OK for some requests, is this the expected behavior or a bug?

Oops, that's a bug at our end.  Some test code got used there.  We just restarted the RTK2go node with another build, so it is gone now.  I do not think it would be a bug to most Rev1 clients, but it has been removed.

AndrewR-L, 2020-03-01:
@nebkat 
Yes the 104 error may be symptom rather than a cause because more often than not the UART time-out happens withiut it.

I am not exactly sure how much information you can get from a coredump, but now that it is working successfully I wonder whether it is worth (at least temporarily) making the UART timeout trigger the saving of a coredump if examination later might give a clue as to which processes/interrupts are in what state at the time to see why XBee is not listening for F9P?

Is a coredump saved when you press the reset button on the SRTK2B (or only on a crash)?

AndrewR-L, 2020-04-04:
I have not been monitoring the NTRIP server log for a few weeks. 

But I have been using the Rover on the tractor and have not noticed any particular losses of fix due to the server resetting/reconnecting.

But having said that I have been forced to use RTK-float mode, because the steering computer wants me to spend $$$ on an "activation code" before it will accept RTK-fixed quality coordinates.

But I see it is still doing panic resets all by itself - eg at about 4am yesterday morning as attached.
[esp32_xbee_v0.5.2_core_dump_dd4176_2020-04-04_07_02_0.zip](https://github.com/nebkat/esp32-xbee/files/4432237/esp32_xbee_v0.5.2_core_dump_dd4176_2020-04-04_07_02_0.zip).

There are now several streams listed as "NTRIP_ESP32_XBee_Ntrip_Server/1.0" and although not the only ones they appear to be reconnecting way above average (@DavidKelleySCSC may have some stats) but it does seem that it is not just me,

DavidKelleySCSC, 2020-04-04:
> There are now several streams listed as "NTRIP_ESP32_XBee_Ntrip_Server/1.0" and although not the only ones they appear to be reconnecting way above average (@DavidKelleySCSC may have some stats) but it does seem that it is not just me,

Is this referring to the RTK2go SNIP NTRIP caster?  I just reset that about two hours ago, so those stats will be a bit flaky in terms of long term connections. [Aside, we are tracking down a memory leak causing that node to be restarted every few days right now].  But here is what I see right now:
>   NTRIP ESP32_XBee_Ntrip_Client/1.0    (4 entries)
>   NTRIP ESP32_XBee_Ntrip_Server/1.0    (20 entries) 

Will gather some more useful stats before the next reset in about 48 hours and post if I can remove the IPs which are privacy details.

I also think that I see in the logs what I think are 'too many' re-connection events from some esp32-xbee devices.  But I had simply thought these were from users who did not have their GNSS in a place where it could track any SVs.   [SNIP drops Base Stations that connect but send no data after ~12 seconds or so] Sorry if I missed it, but has esp32-xbee connection logic been improved such that it will not try to connect unless there is some data to send?  This is a key NTRIP Server issue to avoid causing needless network traffic.

Nothing I recall in recent days has suggested to me that there was anything wrong with the esp32-xbee code.  

> about 4am yesterday

Your local time zone please?

AndrewR-L, 2020-05-03:
Yes that logic was resolved some while ago.

If no data is received for >10 secs then a reconnection attempt with not be made until fresh data arrives (which quite often happens within 20 seconds but by then SNIP has gone)

It seems to me that if you imagine creating an anonymised  spreadsheet with just 2 columns - the NTRIP Agent used by each Mountpoint and the number of reconnections/restarts in any given time period and sorted it by the latter field (descending) you would find several "NTRIP_ESP32_XBee_Ntrip_Server/1.0" in the top 10. Better still group each Agent type and record the average number of reconnections/restarts per site.

You may be right that antenna siting and quality may be an issue but my hunch is that is is not significant,

4am? That would have been from the log so UTC (ESP XBee doesn't do time zones)

PS In Rtk2go it is "Scotland" not "Scottland". Actually you don't need to list Scotland because at least for the time being that is a part of Great Britain which you already listed ....

England + Wales + Scotland = Great Britain
Great Britain + Northern Ireland = United Kingdom

DavidKelleySCSC, 2020-05-03:
> If no data is received for >10 secs then a reconnection attempt with not be made until 
> fresh data arrives (which quite often happens within 20 seconds but by then SNIP has gone)

In fact if the connection has a prior reservation (as most connections with RTK2go.com now do) then the disconnect will not occur for 90 seconds.  The SNIP operator gets a warning after 12.5 seconds, but no disconnect occurs then.  When this sort of thing occurs the sender has accumulated ~4 or ~8k of stale data in its buffer it needs to unload over the connection before 'fresh' data is again sent.  And typically there is some message loss as well. These stale messages are a side effect of the use of TCP/IP, NTRIP Rev2 provide some UDP methods to overcome this, but they are not widely deployed.  So the take-away is that Xbee is working fine by not connecting _before_ there is data and once connected, the SNIP Caster gives it 90 seconds before dumping it.  [I do not known when Xbee will disconnect on its own due to lack of data] 

> Better still group each Agent type and record the average number of reconnections/restarts per site.

Good idea. I have added that to the requested reports list at this end.  

> ...you would find several "NTRIP_ESP32_XBee_Ntrip_Server/1.0" in the top 10

You are right. Looking at the reports we now have, the 'winners' are not all ESP32_XBee devices but way too many are!  Extracted offenders for most re-connections with Xbee include users in these cities

For NTRIP Agent:   NTRIP_ESP32_XBee_Ntrip_Server/1.0
  Located at:   Skrzyszow;   POL
  Located at:   Roland, Canada
  Located at:   Doetinchem, Netherlands
  Located at:   Hollabrunn, Austria
  Located at:   Pultusk, Poland
  Located at:   Temerin, Serbia
  Located at:   Telatyn, Poland
  Located at:   Sochocin, Poland  

But looking at these a fair number seem to "mess up' at first and then have gone on to have connections that persist for hours - I presume these are new users just getting the settings right on RTK2go.  And that is normal.    I am not sure how to be of help on this.  Perhaps a few connections on one of our test serve machines persiting for weeks are now needed?

AndrewR-L, 2020-05-22:
> In fact if the connection has a prior reservation (as most connections with RTK2go.com now do) then the disconnect will not occur for 90 seconds. The SNIP operator gets a warning after 12.5 seconds, but no disconnect occurs then. 

I am not sure of the terminology connections being restarted and reconnected but if I am understanding correctly in a status line such as this reserved one just now ...

` Stream: NR152QB   Up is 04:20:36 Up(29/2) (13:19:56 (H:M:S)  99.6% overall) `

29 is the number of breaks >12.5 secs (is that a lot?) since the last >90 sec break of which there have been 2 since the last time Rtk2go/SNIP came up?

And the last mini-break was 4hr20 previously and the last longer break was 13hr20 previously.

> When this sort of thing occurs the sender has accumulated ~4 or ~8k of stale data in its buffer it needs to unload over the connection before 'fresh' data is again sent. And typically there is some message loss as well. 

Will Xbee I think there is usually no state data because the c20s breaks occur when the XBee seems to forget to listen to what the ZED-F9P is continuously sending. There is data loss but the (also F9P) rover (set to timeout RTK @ the max 60 seconds) fortunately seem not to be upset by this. 

> [I do not known when Xbee will disconnect on its own due to lack of data]

I don't think it does, when data eventually arrives it simply tries to send it to see if the caster is still there - either that succeeds and if so all well and good or it fails in which case it initiates a reconnection.

> Perhaps a few connections on one of our test serve machines persiting for weeks are now needed?

Let me know if this would be useful - I can upload 2 streams from the same F9P simultaneously - one with XBee NTRIP and one with Rtklib STRSVR.

DavidKelleySCSC, 2020-05-23:
AndrewR-L: Forgive me not knowing off hand, but will look at the
      actual code in SNIP in the morning and report back what I find. 
      DCK.  
    
    
    
    On 5/22/2020 4:05 PM, AndrewR-L wrote:
    
    
      
      
        In fact if the connection has a prior reservation (as most
          connections with RTK2go.com now do) then the disconnect will
          not occur for 90 seconds. The SNIP operator gets a warning
          after 12.5 seconds, but no disconnect occurs then.
      
      I am not sure of the terminology connections being restarted
        and reconnected but if I am understanding correctly in a status
        line such as this reserved one just now ...
      Stream: NR152QB Up is 04:20:36 Up(29/2) (13:19:56 (H:M:S)
          99.6% overall)
      29 is the number of breaks >12.5 secs (is that a lot?) since
        the last >90 sec break of which there have been 2 since the
        last time Rtk2go/SNIP came up?
      And the last mini-break was 4hr20 previously and the last
        longer break was 13hr20 previously.
      
        When this sort of thing occurs the sender has accumulated ~4
          or ~8k of stale data in its buffer it needs to unload over the
          connection before 'fresh' data is again sent. And typically
          there is some message loss as well.
      
      Will Xbee I think there is usually no state data because the
        c20s breaks occur when the XBee seems to forget to listen to
        what the ZED-F9P is continuously sending. There is data loss but
        the (also F9P) rover (set to timeout RTK @ the max 60 seconds)
        fortunately seem not to be upset by this.
      
        [I do not known when Xbee will disconnect on its own due to
          lack of data]
      
      I don't think it does, when data eventually arrives it simply
        tries to send it to see if the caster is still there - either
        that succeeds and if so all well and good or it fails in which
        case it initiates a reconnection.
      
        Perhaps a few connections on one of our test serve machines
          persiting for weeks are now needed?
      
      Let me know if this would be useful - I can upload 2 streams
        from the same F9P simultaneously - one with XBee NTRIP and one
        with Rtklib STRSVR.
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/nebkat/esp32-xbee/issues/5#issuecomment-632938777",
"url": "https://github.com/nebkat/esp32-xbee/issues/5#issuecomment-632938777",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    

Regards,
David Kelley
ITS Programs Manager, SubCarrier Systems Corp. (SCSC)
626-485-7528 (Cell)   626-513-7715 (Office)  888-950-8747 (Main)

DavidKelleySCSC, 2020-05-26:
@AndrewR-L:
I had to go look at the actual code to refresh myself on what these numbers precisely meant but I hope the below will explain the report lines a bit more.  It get confusing when you come to understand that the SNIP NTRIP Caster will go out it it's way to keep a connection flowing, making various proprietary allowances for short TCP/IP connection drops and reconnects in the hopes of keeping valid base stations up an running for every second possible.  As stated above, SNIP initiated disconnects for "non reserved stations" (that is, for a station is has no prior history about it), in much short than the 90 sec given to those with reservations.  As  RTK2go is run requiring a reservations (as most SNIP Casters are) , I will only comment about that logic flow. 

```
You quoted this line out of the SNIP::STATUS reply:
Stream: NR152QB Up is 04:20:36 Up(29/2) (13:19:56 (H:M:S) 99.6% overall)

Now, ~3 days later, that same report line now reads:
Stream: NR152QB   Up is 37:04 MM:SS Up (1/38) (37:04 MIN:SEC  99.9% overall) 
      Stream Name  A       B           (C /D)       E              F
```
Notes:
**A**  Some other details such as noting the stream in hidden in the caster table can also appear here
**B** The update is the length of time since the stream's integrity process was last restarted, short pauses (<12s) do to reset this. This is generally when it last connected or re-connected very quickly. 
**C** Is the restart count since that restart (one 1=no restarts over ~37 min and 29 indicated 29 restarts over a 4+ hour period).  A lower number is preferred. 
**D** Indicates the total composite time the stream as been present/active since SNIP application was restarted (in this case, about 8 days ago)
**E**  The percentage time time, since the stream first appeared to the Caster, that it has been active. Most casters are 'stable' in that the same list of Base Stations is generally always present there ecept during problems. RTK2go in an exception in that Bases and NTRIP Clients can come and go at any time for odd periods. As a result a low value here may simply mean that the base was turned off again once it use was completed. 

See also these three charts showing this Base station connection and the data rates (both in and out) for today, last week, and the last three weeks, attached below.  Click to enlarge them. The duration of each is labeled at the bottom left of each chart.  You can see that there has been some additional re connections events occurring in recent days with the NR152QB data stream when compared to the prior weeks.  As to why this increased rate, I have no idea at present. 

[image]
[image]
[image]

> 29 is the number of breaks >12.5 secs (is that a lot?)  [and it could be that the reconnect event occurred much faster than 12 sec)

To me yes that is a lot, something is flaky as you can see looking at the ability to sat connect over the prior weeks.  But at the same time, others have worse and some manage to remain connect without ever a single drop for days at a time.  Unless you can excuse this as a poor WiFi link, I would look into why it occurs. [As always, blame the UART until you are sure] 

>> Perhaps a few connections on one of our test serve[r] machines persiting [persisting] for weeks are now needed? 

>Let me know if this would be useful - I can upload 2 streams from the same F9P simultaneously - one with XBee NTRIP and one with Rtklib STRSVR.

I think it would be a good idea to do this, let me connect with you offline and set you up with an SNIP node that in closer to you (in Paris) that we can dedicate for this need for a few weeks.