nebkat/esp32-xbee #8: Suggest change Wifi hotspot bandwith to 20Mhz

AndrewR-L, 2020-01-13:
Debugging a wifi performance degradation issue I discovered that the hotspot (ESP_XBee_xxxxx) is hogging 8 wifi channels by transmitting with a 40MHz bandwidth (HT40). 

I believe this is the default unless set lower with _esp_wifi_set_bandwidth_ or otherwise in the configuration.

Given that there is never likely to be sufficient throughput to warrant that I suggest this be changed to 20Mhz (HT20) which is the conventional default.

AndrewR-L, 2020-01-18:
Also it would be helpful to have more control over which AP the wifi station connects to on sites where there are multiple APs with the same ssid.

At present I believe the _wifi_scan_method_ remains at the default _WIFI_FAST_SCAN_ mode so in practice (because rssi threshold is not used) it connects to the first AP it finds which matches the selected ssid.

I can see that being good for an RTK base station on a site with a single AP because it would get the RTCM stream up and running more quickly than the alternative _WIFI_ALL_CHANNEL_SCAN_ mode which would necessarily takes longer because all (13) channels would have to be scanned regardless of whether or not there was anything to find.

BUT this behaviour DOES make it very difficult to connect to the strongest signal if there is a weak signal with the same ssid on a lower channel no. No matter how many times I clicked on the top entry (-39dBm)  in the drop down list on the ESP32 XBee Config webpage and "Submit Query" I discovered (eventually!) that XBee was stubbornly connecting to a distant (-83dBm) alternative by the same name because it happened to be on a lower channel number (which increased the number of short term signal outages which XBee need to be able to recover from gracefully.

May I propose either changing the default or including a button to select the search mode between "fast scan" or "all channels" so a new configuation field could be used to populate _config_sta.sta.scan_method_ when configuring the station

Perhaps eventually also the wifi configuration could be extended to show and record the channel number. Configuring the channel in fast scan mode would get the wifi up even quicker as that channel would be searched first, but without being too restrictive as all the other channels would still be searched afterwards if needed (but only if the chosen AP was not found on the selected channel). 

PS I wonder do you need separate _config_ap_ and _config_sta_ structures or whether _wifi_sta_config_t_ is anyway a union of both? NB It works fine as is, just thinking of conservation of memory.

nebkat, 2020-01-27:
> Given that there is never likely to be sufficient throughput to warrant that I suggest this be changed to 20Mhz (HT20) which is the conventional default.

Haven't looked into it before but seems like a reasonable change (c9e2d9cd).

> Also it would be helpful to have more control over which AP the wifi station connects to on sites where there are multiple APs with the same ssid.

Agreed, I've added a switch as you described (5b0b06da). It doesn't seem particularly slow for me even scanning all channels so it should be fine to leave it like this without specific channel choice.

> PS I wonder do you need separate config_ap and config_sta structures or whether wifi_sta_config_t is anyway a union of both? NB It works fine as is, just thinking of conservation of memory.

Currently they are kept to allow the requested config_sta SSID to be given to the /status page even if the connection to that SSID was not successful, but indeed its not necessarily the most efficient. For now I will leave it in case there is other information needed from them in future.

closed 2020-01-28 by nebkat