nebkat/esp32-xbee #9: V0.50 will not load, V0.44 and earlier load fine

n8it, 2020-01-30:
ESP32 module w/128Mbit flash memory.  This ESP32 is installed in a simpleRTK2B board  on the Xbee interface.  UART2 in/out ports have been configured to none.  I have included screen shots of the ESP32 and Flash Download windows along with the terminal output from the ESP32 after loading V0.50 

[image]

[image]

[image]

nebkat, 2020-01-30:
Hi @n8it,

The first 2 screenshots seem to suggest the flashing was successful, and I was able to flash successfully myself. The third screenshot is showing that the device is still in download mode (awaiting new firmware from the PC). Have you tried unplugging the SRTK2B, pressing the XBee Reset button, etc?

n8it, 2020-01-30:
After the download finishes as shown in the first two screenshots, I open the terminal and there is no output.  I then press the reset button on the simpleRTK2B board and then I get the repeating 7 line  blocks until I press the boot button on the esp32 board and that produces the last 2 lines as shown.  I had no issues loading V0.44 without power cycle which is what I did after my problem with V0.50. 
I will try again with a cold start, erase flash, power cycle and follow the wiki line by line.

n8it, 2020-01-31:
I downloaded the 4 V0.50 files again and did a cold start, erase flash, power cycle and followed the wiki line by line I still have the same results as my screenshots.  I have loaded V0.44 again for now.  I will work on this again over the weekend.  Thanks for taking a look at it.

nebkat, 2020-02-03:
@n8it did you have any luck with this? It appears others have been able to flash without any issue, so I'm not sure what exactly is happening in your case. As far as the flashing process is concerned there should be no difference between which exact binaries are being flashed, so its very unusual that v0.4.4 works for you but not v0.5.

AndrewR-L, 2020-02-05:
Just to say I upgraded from 0.44 to 0.5 without any problem.

No idea if it is relevant, but I did have SPI MODE set to "DIO" not "QIO" as above (as in the instructions).

nebkat, 2020-02-10:
If this problem continues with v0.5.1 feel free to comment again but I'll close this for now. There seem to be some isolated cases of this but I don't think its actually related to the firmware.

closed 2020-02-10 by nebkat

n8it, 2020-02-11:
I loaded V0.5.1 on my three simpleRTX2B esp32-xbee devices without any issues.  All three would error out with v0.5.

When I connected all three of my devices to my NTRIP service,  it shows all three Rovers as a  "ESP32_XBee_Ntrip_Client" with Rover Firmware "1.0" .It would be very helpfull if these names could be configured in the esp32 xbee config.

nebkat, 2020-02-11:
Very unusual but I'm glad it appears to have resolved itself.

I'll consider adding an option for that, thanks for the suggestion!