nebkat/esp32-xbee #11: ntripserver adds text

aortner, 2020-02-10:
[image]

if i send data via esp32 to a ntripcaster the esp adds  HTTP/1.1

to the mountpoint

nebkat, 2020-02-10:
What is the caster you are using? The HTTP/1.1 is a standard part of the HTTP request, and a mountpoint can only contain A-Za-z0-9_-. so the portion after the space should be ignored by the caster.

I'm not sure what is in the NTRIP v1 standard document, but in v2 it mentions that the v1 request looks like:

```
SOURCE secret /ExampleMountpoint HTTP/1.1<CR><LF>
Source-Agent: NTRIP ExampleServer/2.0<CR><LF>
<CR><LF>
```

That said if this is a common caster I could add an option to exclude the HTTP/1.1 from the request.

aortner, 2020-02-10:
i am using this: (and i think its the standart caster :-)

https://igs.bkg.bund.de/ntrip/bkgcaster

nebkat, 2020-02-10:
Hmm, if its the professional version they should probably correct the bug, but if its the free version I'll have to add the option to exclude HTTP/1.1.

aortner, 2020-02-10:
i think the bug is on your side... :-)

nebkat, 2020-02-10:
7568ae69 I've removed it for the sake of the old casters but if that is the professional product it should definitely be corrected on their side as its included as an option in the NTRIP v2 specification (for v1 connections).

[image]

aortner, 2020-02-11:
ok - if i unterstand it there are 2 different logins:

ntrip 1 - it uses only password (not encrypted)

and ntrip 2 that uses username and password (encrypeted)

so what do you use or do you use both versions?

i have setup that caster and i like to have ntrip2 login...

nebkat, 2020-02-11:
Currently only NTRIP v1 is supported. There was hope that v2 would be possible using the ESP-IDF HTTP client but that seems unlikely, so I will probably soon implement v2 manually using sockets.

closed 2020-02-12 by nebkat