nebkat/esp32-xbee #15: WPA2-Enterprise

steffnat, 2020-02-20:
Hi nebkat,
I want to setup a base station on the University of Applied Sciences Weihenstephan which has - as lots of Universities - a WPA2-Enterprise secured Eduroam network.
An example of an implementation of the WPA2-Enterprise standard to esp32 can be found here – https://forum.arduino.cc/index.php?topic=556705.0.
Could such a code be implemented? Eduroam or other Radius server access etc. would be great for e.g. universities!
Thank´s!
Steffen

nebkat, 2020-02-24:
Hi @steffnat,
I also have Eduroam in my university, so I'll see what I can do - will mainly have to figure out how to upload and store the server certificate. Until then the best you can do is to use your phone as a hotspot to Eduroam.

steffnat, 2020-02-24:
Hi Nebojša,
first thank's for your excellent work with the WiFi NTRIP Master👍
The method with the phone is clear and I use it for tests but we want to
run a PERMANENT base station and perhaps you know how it is with the public
administration and phones and fees etc. 😩
Did the guy from my provided link (
https://forum.arduino.cc/index.php?topic=556705.0) not solve the problem?
When I can help you in something e.g. testing, German translation or
similar let me know!
Best regards!
Steffen

steffnat, 2020-03-18:
Hi Nebojša,
I hope Corona doesn't affect you and also your country not too much. 
Do you have already an idea how to upload and store the server certificate?
Some estimation about the time schedule for implementation? 
Best regards and stay healthy!
Steffen