nebkat/esp32-xbee #18: Add support SIM800L for NTRIP server

d1ken, 2020-04-04:
Maximum distance between my fields are 60km.
I want to place more base stations on the warehouses near the fields.
But there are only 2G mobile internet.
SIM800L cheap and 2G internet speed for RTCM is enough.

P.S.: Caster has good internet connection and can make NEAR base selection.

nebkat, 2020-04-04:
Hi @d1ken,

I believe there is a 3G module being worked on by ArduSimple, but this GitHub project is just for the ESP32 module. I can also suggest the https://www.digi.com/products/embedded-systems/digi-xbee/cellular-modems/digi-xbee-cellular-3g which can be programmed to work as an NTRIP server.

closed 2020-04-04 by nebkat

d1ken, 2020-04-05:
We use ZED-F9P because of it low price, the same thing going with cellular connection.
Price for Digi xbee cellular 3g is $89 (half price of ZED-F9P), price for SIM800L is 2$.

If I found someone on freelance who can write code for SIM800L, did you accept pull request?
Maybe there are some requirements or restrictions for code?

jmcdrone, 2025-09-17:
? How to add sim 800l to this project? Is it possible?