nebkat/esp32-xbee #19: esp32 ntrip master no connection to the Configuration page

GeorgLinsberger, 2020-04-04:
I have a simpleRTK2B with an esp32 xbee module since this week.
At the beginning I was able to connect to the esp32 WiFi-Hotspot and browse to the Configuration Page. But after a short time it didn't work anymore. I cannot browse to the Configuration Page (ip: 192.168.4.1). I have now reflashed the esp32 (v0.5.2) - that worked perfektly. However, this did not solve the problem. I can connect to the esp32 WiFi hotspot, but I cannot browse to the configuration page. Can you help me please... 

thanks, and best regards Georg

nebkat, 2020-04-04:
Hi Georg,

Did you perhaps accidentally modify the default IP address of the ESP32 for its own hotspot? The device you are connecting with should show you the default gateway somewhere in the WiFi status/settings - it may have changed from 192.168.4.1. Have you also attempted to perform a [full reset](https://github.com/nebkat/esp32-xbee/wiki/Getting-Started#full-reset).

Best Regards,
Nebojsa

GeorgLinsberger, 2020-04-05:
[image]
[image]

Thanks for the fast respond.
No, if I look under "ipconfig / all", I sometimes get the correct default gateway and the hotspot also assigns me an IP address from the correct address space.
Still, I can't browse to the configuration page.
Sometimes I am not shown a standard gateway and I get an IP address from another address space (see pictures). 

Best regards, 
Georg

AndrewR-L, 2020-04-07:
Georg,

Are there other wifi hotspots in range of the windows machine you are using?

In the second example it seems to have connected to an entirely different hotspot.

With some machines if you connect to a hotspot which has no internet access (and of course ESP_Xbee_***** has none) it will actively seek out and connect to another hotspot (even if the signal is less good).

Is it possible that you are telling windows to connect to ESP_XBee_***** hotspot and it does so, but only temporarily and before you have had time to navigate to 192.168.4.1 windows has already found a "better" AP which is not what you want?