nebkat/esp32-xbee #21: WiFi Ntrip master is restarting every 3 second

mlorenz650, 2020-04-06:
As you can see in the terminal, the Nrtrip master is restarting every three seconds. I tried a full reset with the boot button, but it dosen´t work. Because of the restarts, is not providing it´s hotspot, so I can not program it. I have tried the versions 0.5.2, 0.4.2 and 0.4.4 , nothing helps. What can I do?

Thank you for your help!

[image]

arwooldridge, 2020-04-20:
For what its worth I got something similar when I opened the tcp socket client in addition to the Ntrip and ssid, did you have too many boxes ticked in the html. there is a warning about it on that page.

jeremitu, 2020-05-16:
Same here. On the first attempt to connect with WiFi. Didn't configure anything special yet.