nebkat/esp32-xbee #22: Custom NTRIP client on simpleRTK2B board

openedhardware, 2020-04-12:
I do like to create a custom NTRIP client in Python without ESP32-XBee Master device. 
- Which UART is the ESP32-XBee using now? Same UART on `POWER+GPS` USB port?
- If so, do I just send GPS data to the NTRIP server and redirect the response from the server to the GPS module thru `POWER+GPS` USB port?

This would be really helpful if people wants to add the RTK feature to the simpleRTK2B board by other Linux-based SBCs like Raspberry Pi or so.

Thanks!

steffnat, 2020-04-12:
Hi,
read here - https://www.ardusimple.com/question/how-to-configure-xbee-bluetooth/ - it is stated from Ardusimple theat "the XBee socket is connected physically to UART2 so it is not possible to reroute it to UART1."
And some more 
+ https://www.ardusimple.com/question/with-the-simplertk2b-can-you-simultaneously-have-a-base-rover-configuration-and-connect-to-you-android-device-via-bluetooth-at-the-same-time/ 
+ https://hillert.blogspot.com/2020/03/ardusimple-simplertk2b-with-long-range.html
Hope that helps!
Happy Easter!
Steffen

openedhardware, 2020-04-12:
Hi, @steffnat 

Thanks for your quick reply!

So just to double-check:
If I disable all protocols on UART2 on u-center, connect `UART1 TX1 => RX2, RX1 => TX2`, and set the power to 3.3V on IOREF (connected the 3.3V out to IOREF), then I can convey the correction messages to the module thru UART2(`POWER+XBEE` port)?

Thanks again for replying on holiday!