nebkat/esp32-xbee #23: Autoconnect to multiple hotspots option?

arwooldridge, 2020-04-19:
I programmed a Huzzah esp32 and works now, although I had to enter TX/RX pins in config.c as the configuration html page seemed to have  no effect on changing those pins.
Anyway I need to change frequently between two hotspots ( two users) and I found setting up using 192.168.4.1 is quite a painful, and somewhat erratic procedure. 
A useful facility would be to auto connect to two or more preset hotspot SSID's / Password's.
So similar to a PC autoconnecting to multiple wifi SSID's whichever one is available.

arwooldridge, 2020-04-20:
So the flow could be:
Use the first items (SSID, pw) in the csv lists.
Check for valid RSSI after say 5 seconds?
if no valid RSSI use the next SSID in the csv lists.
repeat until end of list 
return to first SSID

gdt, 2020-09-09:
I believe this is a very desirable feature.   Even if this is not N SSID/pw pairs, just having 2 would be very useful.  In my case, I would like the device to connect to my phone's hotspot, if present (usually not enabled), and if not, my home SSID.

An example of a device which behaves this way is the Emlid Reach (original version), which has a list of SSIDs.  This runs GNU/Linux so it is just a wpa_supplicant config file.   Once programmed, I am able to use it with my phone's hotspot (to get RTK corrections), and to have it join the home network (to offload logs, etc.).

A further nicety would be to have the device's own SSID (where it creates the network) be active only when none of the configured SSIDs can be joined.  (This is the Reach behavior, and quite useful.)