nebkat/esp32-xbee #24: 431 Request Header Fields Too Large

jrvoice, 2020-04-21:
Hi,

I keep receiving this in my log file, and when the http page refreshes. The full error is:

25 | 19:45:11.669ms | WARN | httpd_parse | parse_block: request URI/header too long
-- | -- | -- | -- | --
26 | 19:45:11.670ms | WARN | httpd_txrx | httpd_resp_send_err: 431 Request Header Fields Too Large - Header fields are too long for server to interpret

I'm using v0.5.2 with NTRIP server configured to work with rtk2go.com.

Without packet capturing I don't think I can get the full header etc, so any advice on how best to achieve this most welcome! 

A great product, will be very handy...

Jon

nebkat, 2020-04-22:
Hi Jon,

Thanks for submitting, this probably means your browser has some very large header that it is sending to the ESP32. If you are using Chrome, could you try:

1. Open the developer tools (F12)
2. Click on the Network tab
3. Refresh the page
4. Click on the first item that shows up in the list
5. Click on the Headers tab
6. Paste here a screenshot or the contents of the "Request Headers" section

Just to check whether this is an unusual header from your browser, or if it is a standard request for which the maximum request header size should be adjusted in the ESP32.

Best Regards,
Nebojsa

DavidKelleySCSC, 2020-04-22:
This might be off topic, but I post it in case it is not, apologies if it is.  If the "Header fields are too long" could be referring to the _reply_ from the NTRIP Caster in case of error it is easy for this to be quite large.  Please be aware the size of a full reply will vary with the number of entries in the Caster table.  A small Caster might be <500 bytes, while a large one can exceed  150kb.  The RTK2go machine states "Content-Length: 35403" just at the moment (the actual header part is much smaller).  If your connection is accepted you get ICY 200, but if there are any issues you get the full table sent back at you. This is as per the RTCM NTRIP spec.

AndrewR-L, 2020-05-22:
Out of curiosity I checked the size of the Headers in my browser's request was sending to get the local web config page. 

434 characters, so not 512 characters but getting a bit close for comfort. It is amazing how verbose things are these days.

nebkat, 2020-05-27:
Fixed in 8f91f4bad0cfb1dd90fc265462db3f1579a771be

closed 2020-05-27 by nebkat