nebkat/esp32-xbee #25: ubx messages to NTRIP or socket

jubru, 2020-04-22:
Hello,

Well done for your development, it's really cool.
Currently I have 1 simpleRTK2B and 1 WiFi NTRIP Master. I easily send RTCM message to rtk2go.com with NTRIP server connection. 
But, Is it also possible to send the UBX protocol messages from F9P to NTRIP/socket server with the Wifi NTRIP MASTER (messages UBX-RXM-RAWX and UBX-RXM-SFRBX) ?
I activated this from u-center:
[image]
and 
[image]
but no UBX data comes out of the NTRIP MASTER

Thank you in advance
Best regards

nebkat, 2020-04-22:
Hi @jubru,

Unfortunately the ZED-F9P doesn't support the UBX protocol via UART2, the option shown in the u-center configuration is simply to make it consistent with UART1 and potential future devices that might support the feature.

You can however disable UART2, and use jumper wires to connect the output of UART1 to UART2, and in your case you would only need to have the TX line of UART1 connected to make this work.

Best Regards,
Nebojsa

closed 2020-04-22 by nebkat

ardusimple, 2020-04-22:
Soon we will launch a new simpleRTK2B with UART1 on the XBee socket, so this will be possible.