nebkat/esp32-xbee #26: Packets needed for Caster

openedhardware, 2020-07-14:
I have configured my device as a server and caster like this:
[image]

But it doesn't work as I have enabled/disabled some NMEA & UBX packets from ArduSimpleRTK2B module.

Could you give me the list of the packets that are needed for the caster?

Regards.

nebkat, 2020-07-14:
Hi @openedhardware,

Are you attempting to connect as a server to the caster also running on the ESP32? You would normally only be using one of these two features. So if you want your data to be accessible on the ESP32's IP address /H2K_SHOP, you would enable the caster as you have in the screenshot and then connect to that using an NTRIP client.

The ESP32 also does not concern itself with the contents of the data stream, it simply passes along what it receives, so the NMEA/UBX shouldn't affect it (either it is sending data or not).

Best Regards,
Nebojsa

openedhardware, 2020-07-15:
Hi, Nebojsa.
Thanks for your quick reply!

Ok, let me enable `caster` mode only and see what the NTRIP client gets from that.

Regards.

openedhardware, 2020-07-16:
Hi, @nebkat 

I have enabled the caster mode only and tried to download RTCM3 packets with the NTRIP client(u-center).
But it doesn't download anything:

```
01:51:02  Reconnection thread started.
01:51:02  Request thread started.
01:51:28  Ending NTRIP session.
01:51:28  NTRIP session already closed.
01:51:28  Start NTRIP session.
01:51:28  Connected to NTRIP server.
01:51:28  Send mountpoint request ()
01:51:38  Connection error received (10035).
01:51:38  ERROR: Connection lost.
01:51:38  ERROR: Authentication failure.
01:51:40  Reconnection thread ending current session.
01:51:40  Ending NTRIP session.
01:51:40  Closing connection to server.
01:51:40  Server connection closed.
01:51:40  Reconnection thread starting new session.
01:51:40  Start NTRIP session.
01:51:40  Connected to NTRIP server.
01:51:40  Reconnection thread sending last request.
01:51:40  Send mountpoint request ()
01:51:43  Reconnection thread ending current session.
01:51:43  Ending NTRIP session.
01:51:43  Closing connection to server.
01:51:44  Server connection closed.
01:51:44  Reconnection thread starting new session.
01:51:44  Start NTRIP session.
01:51:44  Connected to NTRIP server.
01:51:44  Reconnection thread sending last request.
01:51:44  Send mountpoint request ()
01:51:47  Reconnection thread ending current session.
01:51:47  Ending NTRIP session.
01:51:47  Closing connection to server.
01:51:47  Server connection closed.
01:51:47  Reconnection thread starting new session.
01:51:47  Start NTRIP session.
01:51:47  Connected to NTRIP server.
01:51:47  Reconnection thread sending last request.
01:51:47  Send mountpoint request ()
01:51:50  Reconnection thread ending current session.
01:51:50  Ending NTRIP session.
01:51:50  Closing connection to server.
01:51:50  Server connection closed.
01:51:50  Reconnection thread starting new session.
01:51:50  Start NTRIP session.
01:51:50  Connected to NTRIP server.
01:51:50  Reconnection thread sending last request.
01:51:50  Send mountpoint request ()
01:51:53  Reconnection thread ending current session.
01:51:53  Ending NTRIP session.
01:51:53  Closing connection to server.
01:51:53  Server connection closed.
01:51:53  Reconnection thread starting new session.
01:51:53  Start NTRIP session.
01:51:53  Connected to NTRIP server.
01:51:53  Reconnection thread sending last request.
01:51:53  Send mountpoint request ()
01:51:56  Reconnection thread ending current session.
01:51:56  Ending NTRIP session.
01:51:56  Closing connection to server.
01:51:57  Server connection closed.
01:51:57  Reconnection thread starting new session.
01:51:57  Start NTRIP session.
01:51:57  Connected to NTRIP server.
01:51:57  Reconnection thread sending last request.
01:51:57  Send mountpoint request ()
01:51:58  Ending NTRIP session.
01:51:58  Closing connection to server.
01:51:59  Server connection closed.
01:52:16  Ending NTRIP session.
01:52:16  NTRIP session already closed.
01:52:16  Start NTRIP session.
01:52:16  Connected to NTRIP server.
01:52:16  Send mountpoint request ()
01:52:16  Data received from server (186bytes).
01:52:16  Connection error received (0).
01:52:16  Ending NTRIP session.
01:52:16  Closing connection to server.
01:52:16  Server connection closed.
01:52:27  Start NTRIP session.
01:52:27  Connected to NTRIP server.
01:52:27  Send mountpoint request (H2K_SHOP)
01:52:27  Data received from server (15bytes).
```

Do I have to configure something for the ArdusimpleRTK2B module?

P.S. For some reason the web page doesn't load CSS files...
[10.0.0.9.pdf](https://github.com/nebkat/esp32-xbee/files/4930469/10.0.0.9.pdf)

Not sure what is wrong...

Thanks!