nebkat/esp32-xbee #28: NTRIP compliant messages (GGA only)

fisart, 2020-09-04:
I am using the Ardu simpleRTK2B Board in Connection with ESP32 XBEE Board. I am connected to the Berlin SAPOS System (NTRIP Caster). I use the same WIFI connection to store and analyse the GPS Data via the TCP Socket Server. The Berlin authorities have informed me that my NTRIP Client sends further Information to the caster on top of the required NMEA GGA messages which is not allowed. They asked me to disable those messages. I had not been aware that the NTRIP Client on the ESP32 XBEE Board would sent all Messages which are enabled in the U-BLOX FP9 to the Caster. I strongly suggest  a Software change  whereby the NTRIP Client only forwards compliant NMEA GGA messages as defined in the official NTRIP Protocol to the Caster. Many Thanks in advance for considering my proposal

gdt, 2020-09-10:
I am also seeing this issue.   I am connected to the Massachusetts Department of Transportation network.  While they don't complain, or haven't yet, I see  that I am sending 960 bytes/second.   As I understand it, for nearest or VRS (e.g. IMAX), they take one initial position and use it to find a reference station, or pick coordinates for a VRS, and then keep using it.   Thus, it would be good to suppress everything but GGA, but to send only one position at startup.

MichaelEFlip, 2020-12-05:
I have seen this behavior also on my new Ardusimple Board with the ESP32 XBEE. I would suggest the following selectable options:
- Disable all transfers to the NTRIP caster
- Enable sending the GGA message with a selectable interval (e.g. 10 seconds), the Ardusimple Board should be able to send the GGA message with a higher frequency (1 Hz, 5 Hz, 10 Hz). This should be relayed through the socket server, but not to the NTRIP caster
- Enable forwarding of all messages from the UART Port

I will try to find a solution in the code on my own, but I am not sure if I can do it properly. 
Besides that: Thank you for this cool piece of software!

gdt, 2020-12-05:
Your proposal sounds good to me.   It might be enough to just have GGA at 60s interval rather than needing selected.

I suspect the difficulty in implementation will be adding an NMEA parser, but that's not super hard.

MichaelEFlip, 2020-12-08:
You can disable sending anything to the NTRIP caster by doing the following change. This won't work for SAPOS Germany as they expect a position for the calculation of the virtual reference station (VRS). But for other casters this is okay:

```
diff --git a/main/interface/ntrip_client.c b/main/interface/ntrip_client.c
index 1b5eaf3..749a35f 100644
--- a/main/interface/ntrip_client.c
+++ b/main/interface/ntrip_client.c
@@ -48,7 +48,7 @@ static void ntrip_client_uart_handler(void* handler_args, esp_event_base_t base,
     // Caster connected and ready for data
     if ((xEventGroupGetBits(client_event_group) & CASTER_READY_BIT) == 0) return;
 
-    int sent = send(sock, buffer, length, 0);
+    int sent = 0; // send(sock, buffer, length, 0);
     if (sent < 0) {
         destroy_socket(&sock);
     } else {
```

CuNoah, 2021-01-08:
Have read in another Forum, that Sapos Bavaria works only fine when sending back position at least every ten seconds. Is there any change to implement this? Otherwise the EXP32-XBEE won´t work for this users. 

Thanks in advance.

MichaelEFlip, 2021-03-30:
I added a pull request with a config option to activate/deactivate UART forward to NTRIP server or socket server.
Attention: This does not filter NMEA messages, it only blocks/enables all forwarding.

See https://github.com/nebkat/esp32-xbee/pull/39