nebkat/esp32-xbee #29: Hardware setup with esp32 dev kit

haipas, 2020-11-28:
Hello, I would like to build up a RTK Caster with the software. However, I do not have an xbee_esp32, but an ESP32_DEV_KIT_V4 (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html). I can access the web interface without any problems.

My GPS hardware is an ublox C099-F9P application board. There I have set up everything to send data to the UART pins of the Arduino header. To which pins of the ESP32 do I have to connect the GPS? Pins 1 and 3 of the Uart0 on the board labeled TX/RX do not work. Do I also have to connect RTS/CTS pins?

Thanks in advance

nebkat, 2020-11-28:
Hi @haipas,

UART0 is also used by the USB-UART on the board. Are you attempting to do this while plugged in to USB?

RTS/CTS aren't necessary.

haipas, 2020-11-28:
Thanks for the hint with the USB-UART bridge. I removed the resistors R17 and R18 on the board to disconnect the bridge now everything works.

closed 2020-11-28 by haipas