nebkat/esp32-xbee #30: Is UART1, UART2 or pin TX RX choosing working?

lukaszSZ999, 2020-12-01:
Hi

I would like to sure that setting UART communication (UART0, or 1 or 2) is really working. I tried to use UART1 only for GNSS connection and leave UART0 as main admin connection. What I saw was no matter which UART I checked and which PIN RX TX was set, RTCM data flows only over UART0. 

Maybe I did something wrong or latest firmware have not support UART setting yet?

Best regards
Lukasz

j-python, 2020-12-07:
From what I've seen you need to change it in the source code - it doesn't appear to work

MichaelEFlip, 2021-01-26:
I discovered the same problem here and have no solution yet...

MichaelEFlip, 2021-03-19:
Several UART config items were loaded with a wrong data type.
I added a fix in the following pull request: https://github.com/nebkat/esp32-xbee/pull/38