nebkat/esp32-xbee #42: esp32 WIFI not connecting

Jbt82, 2021-06-25:
My esp32 wont connect to rtk2go. The led for wifi blinks twice while the wifi hotspot led blinks ones then fades away. The led for WIFI server doesn't blink at all. 
In config page it says I'm connected to my localnetwork and I get a IP address. 
In the log I get the following error.

ERROR | NTRIP_SERVER | ../main/ntrip_server.c:133 (ntrip_server_task): Could not connect to host
-- | -- | --

What does it mean?

DavidKelleySCSC, 2021-06-25:
Is your IP perhaps blocked by that Caster due to some prior connection abuse?    
Its is listed in this table?  http://rtk2go.com:2101/SNIP::STATUS#banns

Jbt82, 2021-06-25:
No. I connect to the caster earlier today via USB/STRSVR and it worked fine.

DavidKelleySCSC, 2021-06-25:
And does that method still work now, could you check? [Aside, the ban 'count' is based on the IP, the agent, and the mountPt involved, but once a ban occurs, then the IP itself (regardless of the devices used) is blocked] What is the mountPt involved here?

Jbt82, 2021-06-25:
I connected to rtk2go again a couple of minutes ago with USB/STRSVR and it worked fine. 
SWE4x4GPS is the mountPt