nebkat/esp32-xbee #48: Add bluetooth support for obtaining NMEA

gdt, 2022-01-19:
Separately from NTRIP, it would be very useful to enable accessing the NMEA stream over bluetooth.  This would ease integration with various devices, especially software that already has bluetooth support, which seem to be the standard approach for "external GPS".

(While it would also be nice to have a way for a computer with bluetooth to proxy NTRIP, this request is just about access to the NMEA output stream.)

graememk, 2022-03-03:
I also would like this, The ESP32 has bluetooth, It would mean that to have bluetooth and a Ntrip connection wont need 2 extra hardware pieces.

The UART NEMA feed is there, Putting it over bluetooth hopefully is an easy addition.

I feel that I'm out of my depth to try and implement this myself.

edgecase14, 2023-01-24:
There are many Android apps which support Bluetooth, but not BLE GNSS receivers.  Bluetooth support would enable these.  We can't change our company to use a different software, so we must instead find a different module.

Popeye1001, 2023-10-21:
While looking through the code, I noticed some code for bluetooth support. Is it possible to update with bluetooth functions?

gdt, 2023-10-21:
The astute reader will realize that this code is unmaintained; there hasn't been a release going on 4 years.

You may wish to see if the sparkfun code (which is maintained) can run on this device.   I suspect it's not too hard, but haven't gotten around to trying.

jolivart, 2023-10-22:
Although not open source,there's a BT+BLE functional code here, 100% compatible with the hardware:
https://www.ardusimple.com/btble-bridge-hookup-guide/
Here are the instructions to load the new firmware:
https://www.ardusimple.com/ble-bridge-firmware-update/

gdt, 2023-10-22:
@jolivart interesting.  Looks like it can no longer do NTRIP, which means it isn't workable with QField.   Plus, it's very disappointing to see Ardusimple moving away from open source.

graememk, 2023-10-22:
I ended up building a hat with another esp32 on it to do the Bluetooth serial connection from the main UART output. 

In the the Arduino IDE the Bluetooth stack does take up a considerable amount of memory/flash so possibly having everything in the one package isn't feasible.

gdt, 2023-10-22:
It actually works to have both wifi/NTRIP and bluetooth.    https://github.com/sparkfun/SparkFun_RTK_Firmware/

namork, 2023-11-06:
> [graememk](/graememk)

dear graememk. will you share the code? this would be great to develop an eps32 with an oled to have better control...

graememk, 2023-11-06:
> > [graememk](/graememk)
> 
> dear graememk. will you share the code? this would be great to develop an eps32 with an oled to have better control...

Its brutally simple:

`  
#include "BluetoothSerial.h"
#define RXD2 16
#define TXD2 17

  
BluetoothSerial SerialBT;

void setup() {
  // put your setup code here, to run once:
  SerialBT.begin("ArdusimpleGPS");
  Serial.begin(115200);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
}

void loop() {
  // put your main code here, to run repeatedly:
  while (Serial2.available()) {
    char c = Serial2.read();
    Serial.print(c);
    SerialBT.print(c);
    
  }

}`

namork, 2023-11-06:
> Its brutally simple:
>
> ` #include "BluetoothSerial.h" #define RXD2 16 #define TXD2 17
>
> BluetoothSerial SerialBT;
>
> void setup() { // put your setup code here, to run once: [...]

wow, fast and simple. cool!

that should work with any esp32 with bluetooth?

and it’s hocked to rx/tx on the board?

probably it will also work with the mosaicHAT, right?

thanks a lot!!!!!!!!!!

0marraqueta, 2024-01-16:
Hello, thanks for the code, Everything seems to be working correctly but I can't make the corrections through NTRIP, I always have the RTK FIX=SINGLE solution. How should I have the board configured?

windkh, 2025-07-07:
Any news about this?