nebkat/esp32-xbee #63: no rtk corrections with swipos (swiss rtk corrections)

namork, 2024-06-22:
dear all. if i choose rtk2go in the ntrip client settings, i immediately get corrections. if i set it up with working swipos account, the client connects successfully but doesn't get any corrections. we read somewhere that swipos is picky with gga data sent to it's servers. any hints? and did somebody get it working with v2.refnet.ch?

gdt, 2024-06-22:
Do you believe GGA is sent upstream at all?   I would expect many servers that implement some kind of "virtual reference station" to expect GGA.

namork, 2024-06-22:
yes. at least with v0.5.2 gga is send upstream. my problem was, that i didn't set up our ardusimple-mosaic-x5-receiver to send gaa-data upstream. it's working now with a correction age between one and three seconds. still needs more testing...